sinara-hw/Sayma_AMC #1: Sayma AMC v2 schematic review

gkasprow, 2018-12-12:
@jbqubit I've just published schematics and sources of Sayma AMC v2. I figured out that only a few files are necessary to open the design. Remaining >1k files are not needed. This reduces significantly required space on github. For some reason previous versions of Xpedition tool needed some of them.

gkasprow, 2018-12-13:
not yet, I will finish RTM till the end of this week. There are a lot of changes. Today I will post Metlino schematics. I'm just doing cosmetics.

gkasprow, 2018-12-13:
I  found and fixed some critical error. I'm just curious if somebody finds it during review :)

gkasprow, 2018-12-13:
in the meantime I found and corrected a few issues:
- error in JTAG switch control logic
- too high helper DCXO output level in case of LVPECL oscillator used
- @tprzywoz found issue with JTAG net labels

hartytp, 2018-12-13:
Schematics look much nicer than the last design revision. New release of EE?

hartytp, 2018-12-13:
@WeiDaZhang please can do you do a full design review of the WR on Sayma AMC?

hartytp, 2018-12-13:
I had a look at this for things that I should review for Sayma AMC. The only things I can see are:
- WR oscillators and related connections (I will not do the transciever/FPGA clocking, that needs to be done by @sbourdeauducq)
- RTM connector pin out

Let me know if anyone is expecting me to look at anything else.

gkasprow, 2018-12-13:
I played manually with font setting during  pdf export :)

jbqubit, 2018-12-13:
Thanks @gkasprow !

hartytp, 2018-12-14:
> RTM connector pin out

https://github.com/sinara-hw/Sayma_AMC/issues/3

> WR oscillators and related connections (I will not do the transciever/FPGA clocking, that needs to be done by

https://github.com/sinara-hw/Sayma_AMC/issues/5

hartytp, 2018-12-14:
@gkasprow one other thing to consider:
- should we replace the FP SMAs with MMCXs? In the last revision I found it a bit of a pain to tighten SMAs because the density of components was so high
- if there we make this switch then let's move the WR clock input to the FP, using an MMCX for that as well

gkasprow, 2019-02-25:
I updated SMAs with MCX

closed 2019-02-25 by gkasprow