sinara-hw/Sayma_AMC #3: AMC <-> RTM connections

hartytp, 2018-12-14:
- On Sayma RTM v1.0 there are 10 unused diff pairs on the RTM connector. On the AMC these connect to: GTP8RX-GTP10RX and SYNCOUT{1,2}1
- On Sayma RTM v2.0 we will completely remove (i.e. not just DNP) the JESD ADCs. This will free up an additional 11 diff pairs on the RTM that are currently connected to: GTP{0-7}RX; ADC{0,1}_SYNC; FPGA_ADC_SYSREF
- So, assuming I've counted that correctly, that leaves us with 21 diff pairs to play with.
- On Sayma v2.0 we need to connect more IO to the AFE mezzanines for LVDS ADCs etc. I consider this IO to be part of the "core functionality" of Sayma, since localized control loops (e.g. for laser intensity stabilization) are really important in our experiments
- AFAICT it is not possible to use transceiver TX/RX lines as LVDS, so if we want to repurpose the unused GTP RX lines then we will need to move them onto FPGA SelectIO pins (make sure to include a couple of CC/GC pins)
- Side note: I would be (pleasantly) surprised if using 0R resistors to switch these lines between MGT and LVDS will work from a SI perspective, so my assumption is that the choice of MGT v LVDS will not be a matter of population options

Issues that need considering:
1. How many free pins on the FPGA that we can use for LVDS?
2. Are there likely to be other uses for Sayma AMC that would need the GTX lines? e.g. are people likely to want to combine Sayma AMC with a RTM which breaks out these GTX lines to SPFs for use as a WR/DRTIO switch? My guess is that they won't and, in any case, if people do want this kind of switch, it might be easier to use a really simple AMC combined with FMC<->SFP adapters.

Anyway, once we have answers to these questions we can finalize the layout on the RTM connector.

Other notes:
- Let's try to keep as many of the LVDS lines in the same bank as possible to avoid timing issues
- let's try to delay match the LVDS lines
- need at least two CC/GC LVDS lines (e.g. AFE ADCs)
- On the RTM: ensure that routing of LVDS lines is optimized to minimise cross-talk with analog signals (particularly with the DAC RF outputs).
- FAT_PIPE2 is the link to the second MCH slot in the uTCA rack. This will not be used in ARTIQ, so we can steal lines from this.
- see https://github.com/sinara-hw/sinara/issues/414

hartytp, 2018-12-14:
Other comments:
- final pin choice needs a design review by @sbourdeauducq or @jordens to verify that we're doing things sensibly for use with SU-Servo type loops etc
- Assuming that there are banks on the RTM FPGA that work with LVDS signals, we could consider routing some IO to the RTM FPGA and some directly to the AFE mezzanines. Let's wait and see how much IO we have to play with before we make a decision about this

hartytp, 2018-12-14:
Additionally:
- Now that we don't have an HMC7043 on the RTM, what should we do with GT_CLK{1,2}? We could route the 125MHz reference clock from the RTM to the AMC, but it's not clear that would be useful (and in any case, at some point the RTM ref clock should be derived from the AMC via WR so there is no point routing it back again). Or, we could just re-purpose these as additional LVDS IO. This needs to be considered as part of a review of how to drive the transceiver MGTREF pins in Sayma v2.0
- note to self: in the current version of ARTIQ
  - RTM_FPGA_LVDS2 is used for SERWB TX
  - RTM_FPGA_LVDS1 is used for SERWB RX
  - RTM_FPGA_USER_IO is used for SERWB CLK
  - in the new version of ARTIQ, SERWB will not be used, so these lines become general purpose user LVDS IO
  - @gkasprow why is the naming not consistent here? Why is RTM_FPGA_USER_IO not named RTM_FPGA_USER_IO_0?
  - RTM_MASTER_AUX_CLK is not currently used by ARTIQ and I don't think there are any plans to use it in the future. Can we use this as general purpose LVDS for the AFE mezzanines?
  - NB I am judging pin usage in ARTIQ from https://github.com/m-labs/migen/blob/master/migen/build/platforms/sinara/sayma_rtm.py

hartytp, 2018-12-14:
> we can have additional 4 LVDS pairs. They are routed to FatPipe2 because I had not better idea what to do with them

Great!

That's a good start, but I'm greedy and want more :)

e.g. I'd like to be able to put 1 LVDS ADC  on each AFE mezzanine, so two total. Ideally we should have enough IO for 2x (CLK_IN, CLK_RET, 2xSDO) = 8 lines, as well maybe 1-2 extra lines for future use.

-- 

As I said, there are also a few more LVDS lines we can steal like IO_L22P_T3U_N6_DBC_AD0P_66/IO_L22N_T3U_N6_DBC_AD0P_66 (SYNCOUT11) etc.

Let's rename those all USER_LVDS_X or similar (convention: RTM_FPGA_LVDS_X goes to the RTM FPGA, USER_LVDS_X goes to the AFE mezzanines). Let's make sure we've re-purposed as many IOs as possible for general-purpose LVDS links.

> I would simply connect MGT Rx lines to the analog mezznines.

Do you think SI will be okay once they've gone through the FMCs and onto an FR4 PCB?

Anyway, my preference would be to maximise the amount of LVDS IO we have going to the AFE mezzanines, even if that means loosing a few transceiver lines (realistically, I think it's unlikely that anyone will ever use Sayma with a JESD ADC, so these transceivers are unlikely to be used).

Let's start with the question "what is the most LVDS lines we can use on the AMC FPGA" and then decide about transceivers.

> Take into accout that RTM_FPGA_GTP_Rx1 / TX1 are also LVDS.

If they are LVDS why are they named GTP? I thought that transceivers can't be used for LVDS. If they are just normal select IO pins then let's rename them to RTM_FPGA_LVDS_x.

hartytp, 2018-12-14:
Which transceiver lines will be used for RTM DRTIO?

hartytp, 2018-12-14:
One other thing: I think that LVDS links to the AFE mezzanines are more useful than links to the RTM FPGA.

Once we've decided how many AMC <-> RTM LVDS links we have to play with, we can divide them between AFE and FPGA links.

gkasprow, 2018-12-14:
We can also forward AMC LVDS links via RTM FPGA LVDS IOs as well.

gkasprow, 2018-12-14:
We used LVDS lines to connect to RTM FPGA GTP to stream DRTIO. It is possible, but GTP to GTP is much easier to implement.

jordens, 2018-12-14:
Could you file issues for checking individual questions like the suservo compatibility thing above?

hartytp, 2018-12-14:
> Could you file issues for checking individual questions like the suservo compatibility thing above? And assign to me.

Yes, apologies for the mega issue. Will break out later this weekend.

hartytp, 2018-12-14:
> We can also forward AMC LVDS links via RTM FPGA LVDS IOs as well.

Yes, this was discussed before. We can do this but (a) it's mildly inconvenient since it means changes to the RTM gateware to configure (b) we're currently very short on pins on the RTM FPGA, there aren't enough pins to route all IO via the FPGA (c) for things like fast LVDS ADCs we need to be careful of timings when using FPGAs as buffers.

Essentially, it's possible, but for now I'd prefer to route IO directly to the AFE.

> We used LVDS lines to connect to RTM FPGA GTP to stream DRTIO. It is possible, but GTP to GTP is much easier to implement.

IIRC that was the plan for Sayma v1.0. In Sayma v2.0 DRTIO will be accomplished using transceivers and there will be no DRTIO using LVDS links.

hartytp, 2018-12-17:
In https://github.com/sinara-hw/sinara/issues/564#issuecomment-447679857 @gkasprow said:

> We already have 12 LVDS lines from Sayma AMC FPGA

Which 12 lines are these? I'd like to finalize the RTM <-> AMC connectivity before we start designing the AFE so that I know what we have to work with.

gkasprow, 2018-12-17:
We can  make some of GTP Rx lines double use with capacitor and extend the LVDS count to 12 lines.

jbqubit, 2018-12-18:
>IIRC that was the plan for Sayma v1.0. In Sayma v2.0 DRTIO will be accomplished using transceivers and there will be no DRTIO using LVDS links.

I'm talking with @sbourdeauducq by email about this. He can provide details. DRTIO will be via transceivers on Sayma v2.0.

gkasprow, 2018-12-18:
So far I assigned only 2 LVDS lines to the RTM FPGA and rest of them routed to AFE

gkasprow, 2019-02-25:
I think we can close it.

closed 2019-02-25 by gkasprow