sinara-hw/Sayma_AMC #4: Schematics cosmetics:

hartytp, 2018-12-14:


marmeladapk, 2018-12-17:
I'll add some points from https://github.com/sinara-hw/sinara/issues/209 schematics section that are not resolved:

- [ ] IC40 looks like 0x49 on old RTM schematics (address is changed in AMC table to 0x4B)
- [ ] not zero indexed:
 - uart on top
 - fat pipie (though port0 - port3 may be fat pipe 0)
 - SMA I/O
 - CDR_CLK_CLEAN
 - FMC
 - SFP
- [ ] USR UART not renamed

marmeladapk, 2018-12-19:
I think, that this issue could aggregate issues, that could be fixed after boards are send to manufacturer as these have no effect on layout.

Other cosmetics:

- [ ] Exar GPIO0: general input -> general output?
- [ ] Exar: HW_PWR_good net is used now as enable signal
- [ ] P0V9 should be probably renamed to P0V95 (but net P0V95 already exists, could it be merged? VCCBRAM seems to have low current draw, at least 31 mA is noted on schematics)
- [ ] no notes about softstart settings, reset thresholds, R509 which is DNP and what happens if we insert it,
- [ ] no maximum current noted on molex 12V connector

tprzywoz, 2018-12-20:
JTAG_Configuration:
- [ ] Net label FMC1_PRSNT -> FMC1_PS# (like RTM_PS#)

marmeladapk, 2018-12-21:
- [ ] OSC3 has no value or part number

atcher0, 2018-12-21:
- [ ] USB SERIAL QUAD page is described as sheet 31 of 30; DUAL_IO_TTL_BUFFER page is not numbered
- [ ] OSC4_PWR labeled net is connected to OSC2; OSC3_PWR net is connected to OSC1 
- [ ] many MMC not connected pins are not marked NC
- [ ] ETH2_RGMII bus is not zeo-indexed
- [ ] no notes about IC11, IC12 direction settings, FT4232H hard-wired settings, TPS3808 voltage threshold
- [ ] USB_JTAG bus should be marked as INOUT not OUT (JTAG configuration page)
- [ ] PHY_CFG bus should be marked as INOUT not IN (MDIO - INOUT; MDIO_MODE - OUT) (ETH PHY page)

marmeladapk, 2019-01-07:
- [ ] ADCLK948 Q4 and Si5324 CKOUT1 are not marked NC

tprzywoz, 2019-01-08:
Change position of AMC_ADD and AMC_En net labels
[image]