sinara-hw/Sayma_AMC #5: WR connections

hartytp, 2018-12-14:
- @gkasprow what is the additional cost of making J34 + J35 MMCX instead of UFL? Makes the connections a bit more robust
- @gkasprow why does it say "NOT USED" by IO_L24{P,N}_T3U_N1{0,1}_66? These are used for the recovered clock.
- not important, but on the current Kasli schematics we don't connect to Si5324 INT_C1B. On Sayma you connect that to the AMC FPGA. AFAICT, we don't need this line, so we could drop it to save an FPGA pin.
- currently, the Si5324 CLKOUT1 is not connected to anything. Should we connect that to an FPGA MGTREF pin? Now that we don't have the HMC7043 on the RTM, we don't have a programmable clock source to generate the transceiver clock (due to limitations of the transceiver PLL, we may need different clock frequencies for different JESD line rates). @sbourdeauducq should make a decision on this
- @gkasprow what is AMC_TELECOM_CLOCK?
- replace 100k resistor to ground by J3 with 0R
- thinking about this a bit more, I'm concerned that the decoupling between the decoupling between the various P3V3_CDR supplies. The helper DCXO runs at a different frequency to the master, so any power-supply rail coupling will add close-in spurs onto the master. Similarly, if the Si5324 is used to drive a MGTREF, it could be used to produce a different frequency to the WR PLL. I think the best approach would be to the circuit the way it is, test it carefully when the boards arrive and only add more filtering/LDOs if we find an issue. On the other hand, an extra LC filter is cheap and may avoid issues. What do you think @gkasprow 
- the divider for the main DCXO is not correct. With the connections you have, the ADCLK948 internal 50Ohm termination is enabled, so the 200Ohm source impedance will give quite a low signal. Either disable the ADCLK948 termination (but then be careful about the distance between the DCXO an ADCLK948) or use a smaller source impedance.
- for the helper DCXO please add a note indicating which logic standard and termination should be used. I assume you want this to be used with unterminated LVCMOS? If so, then be careful in the layout to avoid a long length of unterminated trace 

Notes on my review:
- [x] Si5324 connections look good (match Kasli)
- @WeiDaZhang are you happy with the FPGA pin used to drive REC_CLOCK?
-

marmeladapk, 2019-01-07:
Helper DCXO doesn't go to 2,5V bank as indicated in the note, it goes to HP. Output divider should probably also be changed.

marmeladapk, 2019-01-07:
Why does helper DCXO have different short to ground (_N line) than main DCXO?

marmeladapk, 2019-01-07:
IC40 OE should be connected as in IC39 (R383 connected to the left side of R385)

marmeladapk, 2019-01-07:
Should IN_SEL (ADCLK948) have pull-up/down resistor to select default?

tprzywoz, 2019-01-09:
As @marmeladapk said Helper DCXO go to 1.8 V bank, not 2.5 V. 
Also Helper has both: DC coupling and DC bias voltage on FPGA input.

gkasprow, 2019-02-25:
>     * @gkasprow what is the additional cost of making J34 + J35 MMCX instead of UFL? Makes the connections a bit more robust

the same, done

> 
>     * @gkasprow why does it say "NOT USED" by IO_L24{P,N}_T3U_N1{0,1}_66? These are used for the recovered clock.

fixed

> 
>     * not important, but on the current Kasli schematics we don't connect to Si5324 INT_C1B. On Sayma you connect that to the AMC FPGA. AFAICT, we don't need this line, so we could drop it to save an FPGA pin.

It's already routed.
> * currently, the Si5324 CLKOUT1 is not connected to anything. Should we connect that to an FPGA MGTREF pin? Now that we don't have the HMC7043 on the RTM, we don't have a programmable clock source [...]

we use HMC7043. 
Shall I connect it to CLK1 of bank 226?

> 
>     * @gkasprow what is AMC_TELECOM_CLOCK?
It is a clock delivered by MCH. In our case, it is delivered by WR  timing module that sits on the MCH.
> 
>     * replace 100k resistor to ground by J3 with 0R

done

> 
>     * thinking about this a bit more, I'm concerned that the decoupling between the decoupling between the various P3V3_CDR supplies. The helper DCXO runs at a different frequency to the master, so any power-supply rail coupling will add close-in spurs onto the master. Similarly, if the Si5324 is used to drive a MGTREF, it could be used to produce a different frequency to the WR PLL. I think the best approach would be to produce the circuit the way it is, test it carefully when the boards arrive and only add more filtering/LDOs if we find an issue. On the other hand, an extra LC filter is cheap and may avoid issues. What do you think @gkasprow

I added already L18 and L19 and 4.7uF || 1nF caps
We don't need low-frequency filtering, do we?

> * the divider for the main DCXO is not correct. With the connections you have, the ADCLK948 internal 50Ohm termination is enabled, so the 200Ohm source impedance will give quite a low signal. Either [...]

changed to 49R9

> 
>     * for the helper DCXO please add a note indicating which logic standard and termination should be used. I assume you want this to be used with unterminated LVCMOS? If so, then be careful in the layout to avoid a long length of unterminated trace

True, we use 1.8V LVCMOS. That's why 150R + 200R divider was used.
But it has a roughly 80Ohm output impedance.
The diff pair with one line grounded would have similar impedance, so that should work.

>

gkasprow, 2019-03-05:
I think we can close it.
The free clock output of Si5324 has other phase relationship than currently used one.
Moreover, we have 2 clocks connected between and bank 225 and 224
I also routed another copy of CDR clock to bank227 to enable clocking of all MGTs.

closed 2019-03-05 by gkasprow

sbourdeauducq, 2019-03-05:
>  The free clock output of Si5324 has other phase relationship than currently used one.

Yes, it is important that all clocks should be derived from the *same* Si5324 output. Except in very special circumstances, the two outputs have a random phase relationship with each other after each Si5324 reboot.