sinara-hw/Sayma_AMC #6: Servo AMC<->RTM connectivity requirements

hartytp, 2018-12-17:
Let's make sure that the AMC<->RTM connectivity is designed to support a SU-Servo type application, using the AFE ADCs to feedback onto the RF amplitude/phase/frequency.

NB software/gateware development is not funded for this. This issue is just about making sure the hardware can support this

jordens, 2018-12-18:
That would be great.
If the ADC is the same as on Sampler and is connected to the AMC FPGA, make sure that CLKOUT is a CC pin and that the SDO[ABCD] pairs are on the same bank as that pin. And signal levels and SI.
A sufficiently thorough design review and sign-off of the feasibility of the ADC use cases needs more discussion and specification. And funding.

hartytp, 2018-12-18:
Thanks for the clear summary @jordens 

Given M-Labs current limited resources for design reviews, shall we assign checking this to Creotech / @marmeladapk. Reiterating the above, the things to check are:
1. we ideally need 12 LVDS links from the AMC FPGA to the RTM AFE mezzanines (2x{CLKIN, CLKOUT, SDO{A-D}}). It's probably okay if less IO is available, but it needs to be flagged during the high-level review and discussed
2. Check that all LVDS IO for each mezzanine is connected to the same bank of the AMC FPGA and that at least 1 CC pin is routed to each AFE mezzanine
3. Verify SI from the AMC FPGA to the AFE mezzanine is good for DDR signals at 125MHz.

jbqubit, 2018-12-18:
Thanks for the summary @hartytp!!

marmeladapk, 2019-01-28:
It seems that we have enough pins, however checking bank connectivity on AMC needs to wait until #34  is resolved.

gkasprow, 2019-03-06:
We have more LVDS than we need.

closed 2019-03-06 by gkasprow