sinara-hw/Sayma_AMC #9: BOM verification

atcher0, 2018-12-17:
BOM: https://octopart.com/bom-tool/L1u7kYdF

Elements out of stock:
- N25Q256A13ESF40 (256Mb, 2.7 to 3.6V, Multiple I/O Serial Flash Memory) - IC1-IC3

Elements with low avability / near end of life:
- BSH103 (30V 850mA N-Channel Enhancement Mode Field-Effect Transistor) - T1-T6
- 549CBAC000112ABG (Ultra Low Jitter I2C Programmable XO (95 fs), 0.2 to 250MHz) - IC39,IC40
- 62101021021 (WR-PHD 1.27 mm SMT Dual Pin Header, H=3.80 mm) - J36
- MT41J256M16HA-125:E (DDR3 SDRAM) - IC22-IC27
- 342947 (COPPER HEATSINK 57.9X59X11MM) - HS1
- XCKU040-1FFVA1156C (Kintex UltraScale FPGA, 520 I/O, 530K Cells, 1156-Ball BGA, Speed Grade 1, Commercial Grade) - IC37

Notes:
- I assumed that 88E1512-XX-NNP2I000 is the same as 88E1512-A0-NNP2I000 (ETHERNET PHY)
- TCM2-43X+ is available only directly from Mini Circuits

jbqubit, 2018-12-18:
Thanks @atcher0! I'm inquiring with Xilinx as to why XCKU040-1FFVA1156C has such poor availability.

gkasprow, 2018-12-18:
Large FPGAs have usually large lead time and are assigned and delivered on per-project basis.

atcher0, 2018-12-20:
- [ ] include FMC standoffs and screws in BOM
- [ ] include SMA insulating washers in BOM

gkasprow, 2019-02-25:
@sbourdeauducq @jordens I will use 
MT25QL256ABA8ESF-0AAT  as FLASH
and MT41K256M16TW-107:P.

marmeladapk, 2019-04-12:
TPS53353DQPT doesn't have manufacturer part number.

closed 2019-04-28 by gkasprow