sinara-hw/Sayma_AMC #14: FP connections

hartytp, 2018-12-20:
Since we're now using MCX elsewhere, can we switch the MMCX connectors on Sayma AMC for MCX please (assuming there is room)? Thanks and apologies for the last minute switch!

jbqubit, 2019-01-09:
@gkasprow @marmeladapk Can this change be accommodated? Seems like it's a trivial change but it does modify the BOM.

hartytp, 2019-01-09:
To clarify, on v1.0 there were 2 SMA connectors on the FP. That didn't really work since they clashed with the USB connection and uTCA handle. This made them a PITA to use.

In the initial v1.2 schematic they were replaced with MMCX to solve these mechanical issues. Proposal here is to replace with MCX...

gkasprow, 2019-01-09:
We have more place with FMC panels so it should be easier.

gkasprow, 2019-01-09:
ups, U are talking about AMC..true, we can replace with MCX ones

hartytp, 2019-01-09:
Yes, I mean on the AMC. Those SMAs always made me swear every time I used them.

I think it's also worth replacing the MMCXs on BaseMod with MCXs where possible to improve the mechanical reliability and compatibility with other boards, but that's a separate issue.

gkasprow, 2019-02-25:
MCX won't fit on BaseMod,
I replaced SMAs with MCXs on the AMC.

closed 2019-02-25 by gkasprow