sinara-hw/Sayma_AMC #19: [needs checking] GTH quad number clock constraints

jordens, 2018-12-20:
With the clocking tree I am a bit uncertain which clock is doing what currently. @hartytp @sbourdeauducq could you confirm the rolses of the clocks?

* The clock that will drive the JESD GTHs should be central in the 4 JESD quads. If it's GT_CLK2_IN_N and GT_CLK1_IN_N the assignment is ok.
* The clock for SFP/AMC also needs to be within 2 quads of the SFP/AMC quad. If all their clocks are on Quad 224, that's ok. Only GT_CLK1_IN_N can clock Quad 224, GT_CLK2_IN_N can not.

[image]

jordens, 2018-12-20:
https://github.com/sinara-hw/Sayma_AMC/issues/8#issuecomment-449038059

closed 2018-12-20 by jordens

hartytp, 2018-12-21:
> With the clocking tree I am a bit uncertain which clock is doing what currently. @hartytp @sbourdeauducq could you confirm the rolses of the clocks?

@jordens I think this one is best left to you and @sbourdeauducq since you know far more about transceiver clocking than me.

FWIW, my thoughts on your comments would be:

>  If it's GT_CLK2_IN_N and GT_CLK1_IN_N the assignment is ok.

Currently there is no plan to add a synth on the RTM to drive those. We could drive them directly from the RTM reference clock, but I don't think that works for all line rates due to the FPGA PLL limitations.

So, I think we should probably remove these signals from the AMC<->RTM connector and drive them from the AMC directly (e.g. from the spare Si5324 output). But, this decision should be made by someone whose thought more about JESD clocking than I have.

jordens, 2018-12-21:
I'm confused. Isn't the clock clocking concept designed already? I am just asking for an explanation of the intended function and mode of operation of a few of the signals going into and coming out of the clock tree.

hartytp, 2018-12-21:
@jordens to clarify:
- the RTM clock tree is fully designed already as part of the new synchronisation scheme work I did
- when @sbourdeauducq and I discussed the new synchronisation scheme we agreed that the transceiver clocking on the AMC would need some modification because of the lack of the HMC7043. IIRC the conclusion of those conversations was that this shouldn't be a major issue, but it needed some thought, and maybe a new PLL on the AMC (unless we can just use the second output of the Si5324). IIRC we agreed to put thinking about this off until M-Labs had funding for it
- AFAICT, our options are either to use the AMC CDR clock + a PLL like the Si5324, or to use the RTM reference clock (routed via the AMC<->RTM connector) + PLL.

hartytp, 2018-12-21:
>  think there needs to be a CDR. Otherwise there is way to much dependency on the RTM and a correct clock coming from RTM

Yes, I think there needs to be CDR (Si5324 + WR) on both the RTM and AMC.

> We have to make sure that we can bring up and RTIO-clock the RTIO stuff on the AMC without the RTM.

The clocking for Sayma v1.0 was a bit of a mess of interdependencies between the two boards. Hopefully we can do something simpler for v2.0