sinara-hw/Sayma_AMC #30: DDR64 clocking with MIG

kaolpr, 2019-01-10:
There is an issue when Xilinx MIG is to be used instead of MiSoC DDR core for DDR64. When one wants to use MIG there is no other way but to provide clock signal using UFL connector. 

It is so because according to the "Clocking" sections of (PG150) _LogiCORE IP UltraScale Architecture-Based FPGAs Memory Interface Solutions_ clock source must be placed in the same IO column as memory interface and no external MMCM can be used (as cascading MMCMs is not allowed and another MMCM is instantiated in MIG).

The problem could be solved by adding 200 MHz oscillator and connecting it to one of the banks in the memory interface column (44 - 48). It may require moving UFL connector elsewhere as I don't see any free differential clock-capable inputs in those banks.

gkasprow, 2019-01-10:
We can also route unused reference buffer outputs to DDR clock inputs.

kaolpr, 2019-01-10:
We can safely assume that this signal will be 125 MHz in all cases?

gkasprow, 2019-01-10:
All oscillators are programmable so we can assume it, however in ARTIQ we use either 100 or 150MHz

gkasprow, 2019-02-25:
added 2 clock paths
[image]
[image]

[image]

closed 2019-02-25 by gkasprow

sbourdeauducq, 2019-02-26:
So this clock gets replaced with the CDR clock, correct?
```
    ("sysclk1_300", 0,
        Subsignal("p", Pins("F18")),
        Subsignal("n", Pins("F17")),
        IOStandard("DIFF_SSTL15_DCI"), Misc("OUTPUT_IMPEDANCE=RDRV_40_40")
    ),
```

gkasprow, 2019-02-26:
Yes.