sinara-hw/Sayma_AMC #33: Clk_recovery LDO

tprzywoz, 2019-01-16:
There should be added another LT3045 (P3V3_CDR) because we exceed output current limit (500 mA).
Worst case for Si5324 + ADCLK948 + 2 x 549CBAC = 863 mA.

gkasprow, 2019-01-16:
ADCLK current depends on a number of loaded outputs. I will check my current estimation

gkasprow, 2019-02-25:
Let's kick away the LDO.
I will add a dual stage LC filter and connect the whole clocking circuit to the P3V3.
When we had only Si5324 I used 5V helper buck (for Exar) to supply it.
Now we have much more stuff- 2 oscillators and ADCLK948. We would have to use a higher current buck and higher current LDO.
Any objections @hartytp ?

hartytp, 2019-02-25:
@gkasprow so the suggestion is to clock all clock recovery components from the P3V3 SMPS? I'd really prefer not to do that if we can avoid it, as I'd worry about coupled noise.

Can we keep an LDO for this revision? Maybe also add a solder jumper that can be used to bypass the LDO, so we can measure the performance with and without the LDO and then consider scrapping it in the next revision once we have some performance data?

hartytp, 2019-02-25:
As you say, a lot of the current estimate for this comes from unused channels on the ADCLK948. The actual worst-case current consumption will be a lot lower than 863mA won't it?

gkasprow, 2019-02-25:
The unused channels don't consume large current until you load them with termination or pulldown.
OK, I will place a stronger buck converter. It is also a lower cost one.
Not worth risking.

closed 2019-02-25 by gkasprow

gkasprow, 2019-02-25:
I used [TPS78633](http://www.ti.com/lit/ds/symlink/tps786.pdf)