sinara-hw/Sayma_AMC #37: Sayma clocking write up

hartytp, 2019-02-23:
- [ ] post a write up + block diagram of Sayma AMC+RTM clocking

hartytp, 2019-02-23:
@sbourdeauducq @WeiDaZhang if you have any thoughts about the best way of getting the recovered clock out of the RTM (Artix) FPGA and into the DDMTD input then please let me know.

sbourdeauducq, 2019-02-25:
On Artix-7 there is no OBUFDS_GTE3 so we have to go through the fabric (and live with the noise).
The path through BUFMR followed by either BUFIO or BUFR may or may not be lower-noise. Before PCB sign-off we should try a Vivado compilation with this path and the actual proposed pin assignments, and check that it does not complain, so we leave the BUFMR option open in the hardware.

gkasprow, 2019-03-11:
I will ask guys from Creotech to draw something similar as they did for the AFCZ. They will update the documentation anyway.
[image]

hartytp, 2019-03-11:
Wonderful, thank you! I started, but it's a lot to do and I didn't finish yet.

hartytp, 2019-03-13:
Keeping track of clock signals:
- `AMC_TELECOM_CLOCK`
  - what is this, and what would we use it for?
- `EXT_CLK_IN`
  - diff input via UFLs
  - as an alternate variant, solder jumpers can be used to connect to either telecom clock B or `WR_REF_CLK_IN`
  - goes to GC input in bank 47
  - what is this signal for? AFAICT it's just for debugging
- `WR_REF_CLK_IN`
  - single-ended input (MCX), balun (TCM2-43-X+)
  - goes to bank 66 with other WR signals
  - what is the use-case for this? AFAICT it doesn't go to the front-panel so it's only intended for debugging?
- `REC_CLK_C`
  - recovered clock from the FPGA transceiver (bank 48)
  - clocks the SI5324
  - can also be used as the reference input to the DDMTD DFF by component options/solder jumpers
  - this will be our WR reference clock if we find any issues when using `MGT_CDR_CLK`
- `MGT_CDR_CLK`
  - FPGA transceiver clock output from MGTREFCLK1
  - CML output, which we hope will have extremely low noise (needs testing)
  - clocks the DDMTD DFF
  - our plan A will be to use this as the WR clock
- The recovered clock (main DCXO or Si5324) is buffered and produces the following signals
  - `CLK_CDR_CLEAN0` - goes to a GC pin in bank 66 (NB this can be used for DDMTD if we don't use the external DFFs)
  - `CLK_CDR_CLEAN1` - goes to `MGTREFCLK0` to clock transceivers
  - `CLK_CDR_CLEAN2` - `MGT_REF_CLK0` to clock transceivers
  - `CLK_CDR_CLEAN3` - `MGT_REF_CLK0` to clock transceivers
  - `SDRAM32_CLK` - goes to a GC pin in bank 68 (`DDR_32_CLK`) will we actually use this to clock the SDRAM? 
  - `SDRAM64_CLK` - goes to a GC pin in bank 45 (`DDR_64_CLK`) will we actually use this to clock the SDRAM?
  - also available on UFL connectors
- `HELPER_DCXO`
  - the helper oscillator for the WR loop. This is frequency locked at a small offset from the main DCXO to provide a WR frequency of a few kHz to get very low jitter for the phase detector.
- `DDMTD_OUT_CDR_CLK_SSTL18`
  - single-ended output signal from the DDMTD DFF (the transceiver recovered clock sampled by the DDMTD DFF)
  - routed to an input in bank 66 with other WR signals (we don't have enough pins for a diff input!)
  - with different population options, this can be routed as a differential signal to bank 68 as `DDMTD_OUT_CDR_CLK`. We'll try to avoid this to keep all WR signals in the same FPGA bank
- `DDMTD_OUT_Helper_CLK_SSTL18` same as above but for the main WR DCXO 
NB @sbourdeauducq the DCXO oscillators are enabled by default so make sure to pull the OE low in gateware when not using them

sbourdeauducq, 2019-03-14:
> * `SDRAM32_CLK` - goes to a GC pin in bank 68 (`DDR_32_CLK`) will we actually use this to clock the SDRAM?
> 
> * `SDRAM64_CLK` - goes to a GC pin in bank 45 (`DDR_64_CLK`) will we actually use this to clock the SDRAM?

That's just to make the pinout compatible with the Xilinx core, MiSoC/ARTIQ will not use it.

hartytp, 2019-03-14:
overall plans for AMC clocking:
- assuming we only run Sayma as a DRTIO satellite
- RTIO reference clock comes from master via DRTIO link
- plan A is to use the Si5324 for clock recovery
- plan B is to use WR
- FPGA fabric clock from `CLK_CDR_CLEAN0` 
- for WR we use either `MGT_CDR_CLK` or `REC_CLK_C` as the reference clock for the DDMTD
- WR core 
@sbourdeauducq I think that covers all the important points for the AMC clocking, right? (Obviously, a BD would still help)

gkasprow, 2019-03-15:
@hartytp 
AMC_TELECOM_CLOCK - it is used in MTCA crate to synchronize multiple cards
EXT_CLK_IN - somebody wanted to have the WR clock input also GC-capable. SInce WR_REF goes to another bank where there are no free GC inputs, I routed it also as an option to the bank where GC pair is available.
WR_REF_CLK_IN - can be placed on a front panel.

hartytp, 2019-03-16:
> AMC_TELECOM_CLOCK - it is used in MTCA crate to synchronize multiple cards

Is there any potential use for this in ARTIQ?

hartytp, 2019-03-16:
> WR_REF_CLK_IN - can be placed on a front panel.

If there is space that would be nice. Good to have a high-quality clock capable input on the FP. If not then no problem.

gkasprow, 2019-03-16:
> 
>     AMC_TELECOM_CLOCK - it is used in MTCA crate to synchronize multiple cards
> 
> Is there any potential use for this in ARTIQ?

not really. These can be essential in the systems where WR receiver that sits on MCH is used to synchronize the AMC boards.

hartytp, 2019-03-16:
On the RTM side, the clocks we have are:
- `GT_CLK0_IN`
  - HMC7043 output routed to AMC. This clocks the AMC JESD204B transceivers
- CDR CLEAN clocks are all copies of the recovered clock from the SI5324/WR DCXO
- `CDR_CLK_CLEAN0` and `EXT_100M_RF_IN`
  - reference clock for the HMC830 and AFE mezzanines (mux selection)
- `CDR_CLK_CLEAN1`
  - goes to RTM FPGA MGTREF. Clocks DRTIO transceivers?
- `CDR_CLK_CLEAN2`
  - goes to MRCC pin on RTM FPGA (bank 34). Main ARTIQ fabric clock for RTM.
- HMC7043 outputs. I'm not clear on which of these are used and how in ARTIQ these days.
- `AMC_FPGA_SYSREF0_P`
- `AMC_FPGA_SYSREF0_P`
- `RTM_FPGA_SYSREF0`
- `RTM_FPGA_SYSREF1`

Also, why is is `REC_CLK_C` not just `REC_CLK`?

gkasprow, 2019-03-16:
> Also, why is is REC_CLK_C not just REC_CLK?
_C means that it is the same signal but after the coupling capacitors.

hartytp, 2019-03-16:
aah, okay, thanks for clarifying

sbourdeauducq, 2019-03-21:
> `MGT_CDR_CLK`
> 
>     * clocks the DDMTD DFF

It does't clock it. It is sampled by it. The FFs are clocked by the helper oscillator.

closed 2019-09-30 by hartytp