sinara-hw/Sayma_AMC #39: Ethernet PHY programming

sbourdeauducq, 2019-02-25:
I wonder if it is really necessary to allow connecting the PHY MDIO (as well as MODE and RESET) to the MMC. The FPGA should be able to program it and handle MODE/RESET, and the MMC does not need Ethernet.

gkasprow, 2019-02-25:
true, the PHY needs switching from 1000base-T to 1000base-x mode by setting one bit. By default it talks to the RJ45 connector. For diagnostic purposes, I added pin header for such a connector.
If you can handle this, I can scrap the mux.

sbourdeauducq, 2019-02-26:
Yes, scrap the mux. What register(s) should be written exactly?

sbourdeauducq, 2019-02-27:
In https://github.com/jbqubit/saymapp/blob/master/pp.md#ethernet
"Configuration of M88E1512 is by static pin strapping: 1 Gbps, GMII, 1000BASE-X."
Is that incorrect?

gkasprow, 2019-02-27:
not really. Look [here](https://www.marvell.com/documents/eoxwrbluvwybgxvagkkf/) p39
One needs to set one bit in register
[image]
That's why I wanted to do it via MMC so you don't have to talk to the PHY.

sbourdeauducq, 2019-02-27:
Doing it in the FPGA is certainly easier and less frustrating.

gkasprow, 2019-02-27:
OK, so let's scrap the mux and close it.

gkasprow, 2019-02-27:
do you need the PHY reset to be routed to the FPGA?

gkasprow, 2019-02-27:
and what about the INT?

sbourdeauducq, 2019-02-27:
I need the PHY reset (always useful to recover from problems), INT is not necessary but can be routed if easy.

gkasprow, 2019-02-27:
OK, managed to connect both.

closed 2019-02-27 by gkasprow