sinara-hw/Sayma_AMC #42: ALT_PORT0 must be usable at the same time as RTM_FPGA_GTP

sbourdeauducq, 2019-02-25:
* RTM_FPGA_GTP is required for DRTIO with the RTM. We actually plan to use DRTIO *exclusively* in the future for communicating with the RTM.
* ALT_PORT0 is the preferred way of using Ethernet; if the GTH behaves itself we don't need an external PHY chip.

Since we are out of transceiver channels, I propose removing one of the two SFPs and allocating it to either ALT_PORT0 or RTM_FPGA_GTP.

Also, the possible connection to FAT_PIPE1 of RTM_FPGA_GTP does not seem necessary, since 
RTM_FPGA_GTP will be a requirement for using the RTM. It makes more sense to make a possible connection to the remaining SFP instead.

sbourdeauducq, 2019-02-25:
If there is an interest in keeping two SFPs, sharing a SFP with ALT_PORT0 is also an option.

gkasprow, 2019-02-25:
done
[image]

closed 2019-02-25 by gkasprow