sinara-hw/Sayma_AMC #44: Sayma AMC v2 design changes

gkasprow, 2019-02-25:
Sayma AMC v2 apart from fixing issues reported in #600 undergoes major modifications:
- Ethernet PHY
- Clock distribution with [WR support](https://github.com/sinara-hw/meta/issues/15)
- FTDI JTAG -copied from Kasli
- JTAG switch - copied from Xilinx devkit, but with AMC access

gkasprow, 2019-02-25:
added clock path between CDR and SDRAM banks to enable Xilinx IP using during production tests
#30

gkasprow, 2019-02-25:
added Silabs CDR test points

gkasprow, 2019-02-25:
added TVS #13

gkasprow, 2019-02-25:
the output of Si5324 buffer routed to the bank 226

sbourdeauducq, 2019-02-26:
> the output of Si5324 buffer routed to the bank 226

Which output of Si5324?
Why?

gkasprow, 2019-02-26:
to make sure every GTX can drive DRTIO

sbourdeauducq, 2019-02-27:
I assume you mean the output of the CDR clock mux/distributor IC44.

OK. Not strictly necessary (the JESD transceivers do not need det-lat with the RTIO clock and also can be clocked from the RTM), but fine to have if it's easy and doesn't break anything.

gkasprow, 2019-03-11:
I think we can close it. It's hard to track all changes since the design is so different from v1.

closed 2019-03-11 by gkasprow