sinara-hw/Sayma_AMC #50: DNP ALT_PORT0 jumpers

jordens, 2019-02-28:
Depending on how you do the routing, you'll probably want to DNP C182-C185.

[image]
[image]

gkasprow, 2019-03-05:
done

sbourdeauducq, 2019-04-10:
GTH Ethernet is working on Sayma v1 now, and we'll probably want to use that (no need to configure the PHY chip and no issues with RGMII timing) - so it's the capacitors for the other path (with the PHY chip) that should be DNP'd.

sbourdeauducq, 2019-04-10:
Do not remove the PHY chip though, just in case there are unforeseen transceiver problems.

gkasprow, 2019-04-27:
@sbourdeauducq in such case we will loose one SFP. Maybe it should be not mounted?

sbourdeauducq, 2019-04-28:
OK.

closed 2019-05-20 by gkasprow