sinara-hw/Sayma_AMC #52: GC pins SMA/I2C/WR_REF

jordens, 2019-03-03:
* Does the I2C interface to the oscillators really need GC pins? If not (and I'd hope that the I2C cores don't), I'd suggest swapping the SMA_IO lines and the various I2C signals so that the SMA_IOs are on GC pins.
* Also the WR_REF_CLK_IN is on a non-GC pin currently. That prevents if from being used for anything other that WR-style phase detection. It would not be usable for general clocking purposes during boot strapping, characterization, or other ideas. May be fine, but needs explicit ack IMO.

[image]

[image]

gkasprow, 2019-03-05:
I can route it to neighboring bank 47 I have there one free GC pair. @hartytp ack please

gkasprow, 2019-03-05:
Anyway if we want to use ext WR clock, shouldn't we deliver it to external DDMTD as well?

hartytp, 2019-03-06:
>  Does the I2C interface to the oscillators really need GC pins? If  not (and I'd hope that the I2C cores don't), I'd suggest swapping the  SMA_IO lines and the various I2C signals so that the SMA_IOs are on GC  pins.

No. It's only a 1MHz interface, so I don't expect any issues here.

jordens, 2019-03-06:
Speed is not the question. Whether or not you clock logic from SDA/SCL is. If you don't oversample, you'll have to.

WeiDaZhang, 2019-03-06:
The implementation of I2C we are using is oversampled, and the SCL/SDA are act as normal 'inout' rather than clock.
https://github.com/ktemkin/VHDL-light-sensors/blob/master/i2c_master.vhd
So,
> Does the I2C interface to the oscillators really need GC pins? If not (and I'd hope that the I2C cores don't), I'd suggest swapping the SMA_IO lines and the various I2C signals so that the SMA_IOs are on GC pins.

No. @jordens 

We ran it in our imploementation of WR on KC705, where the I2C pins are not on GC/CC/SRCC/MRCC neither.
[image]

hartytp, 2019-03-06:
> https://github.com/ktemkin/VHDL-light-sensors/blob/master/i2c_master.vhd

We'll probably want to switch to the [misoc](https://github.com/m-labs/misoc/blob/master/misoc/cores/i2c.py) core for the ARTIQ implementation however.

marmeladapk, 2019-03-06:
@hartytp This misoc core doesn't work. I spent some time in Creotech fixing it, I could sent PR sometime in the future.

hartytp, 2019-03-06:
Good to know! I’m sure bug fixes would be welcome.

gkasprow, 2019-03-06:
The I2C pins were connected to the GC pins by pure coincidence. I did not plan to clock the FPGA fabric using I2C clock :)

gkasprow, 2019-03-06:
I can make assembly options and route WR clock to either IO pins in WR bank or GC in another bank so it can be used as a clock input.

gkasprow, 2019-03-06:
done, closing

closed 2019-03-06 by gkasprow