sinara-hw/Sayma_AMC #58: serwb signals?

sbourdeauducq, 2019-03-11:
On v2.0rc4, the signals used for serwb (e.g. ``rtm_fpga_lvds1_p``) are simply called e.g. ``lvds2``.
Are we sure that they will go to the correct pins on the RTM FPGA?
In particular the clock pins (currently J8/H8 - LVDS11 on the AMC side) must go to a clock-capable input on the RTM FPGA.

It is particularly important that serwb works correctly as @jbqubit refuses to fund a migration to pure DRTIO.

sbourdeauducq, 2019-03-11:
Similary, what used to be a HMC7043 SYSREF signal going to the FPGA is now "LVDS10" (unless I'm mistaken).
This is an important signal (for DAC synchronization) so it must be connected correctly. It does not need to be on a GC pin.

hartytp, 2019-03-11:
> It is particularly important that serwb works correctly as @jbqubit refuses to fund a migration to pure DRTIO.

I'm not sure that's entirely the most constructive way of making your point here.

hartytp, 2019-03-11:
> On v2.0rc4, the signals used for serwb (e.g. rtm_fpga_lvds1_p) are simply called e.g. lvds2.

Yes, that is a change that we discussed on GitHub some time ago. The rationale was that in v1.0 we found that the usage of some of these lines changed after the schematics were frozen. It was confusing to have a mismatch between the schematic and code. So, seems better to avoid encoding functionality into the signal names in these cases.

> This is an important signal (for DAC synchronization) so it must be connected correctly. It does not need to be on a GC pin.

Can you be more specific? What are the requirements for this signal?

> In particular the clock pins (currently J8/H8 - LVDS11 on the AMC side) must go to a clock-capable input on the RTM FPGA.

Is that the only requirement for this signal?

sbourdeauducq, 2019-03-11:
> > This is an important signal (for DAC synchronization) so it must be connected correctly. It does not need to be on a GC pin.
> 
> Can you be more specific? What are the requirements for this signal?

With the current synchronization scheme, that signal goes to an ISERDES input that is clocked by the GTH, and there are no particular requirements.

> > In particular the clock pins (currently J8/H8 - LVDS11 on the AMC side) must go to a clock-capable input on the RTM FPGA.
> 
> Is that the only requirement for this signal?

Yes.
If we use generic names then maybe we can use "CC_RTM" in the name (similar to what FMC does) to indicate that the signal goes to a CC pin on the RTM side.
I think there still should be a difference in names between signals that are intended for the RTM FPGA and those intended for mezzanines.

hartytp, 2019-03-11:
> I think there still should be a difference in names between signals that are intended for the RTM FPGA and those intended for mezzanines.

I don't feel strongly about this. However, I believe we previously agreed not to do that on the grounds that we might end up using Sayma AMC v2.0 with something other than Sayma RTM v2.0, so best to keep all signal names generic from the AMC POV.

Adding _cc for lines that go to a CC pin on the AMC side does sound like a good idea.

gkasprow, 2019-03-11:
The clock capable signals are called _CC on both AMC and RTM side.
You have 3 LVDS pairs that connect both FPGAs.
LVDS25, 26 and 27
LVDS26_CC is routed to GC pins on AMC side (1V8 bank) and MRCC on RTM side (2V5 bank)
[image]
[image]

gkasprow, 2019-03-11:
I've just changed their direction to bidir.

gkasprow, 2019-03-11:
Now there are 4 SYSREF signals connected. Two go to AMC FPGA LVDS21(bank47) and LVDS30(bank66)
Another two go to RTM FPGA SRCC and IO pins

sbourdeauducq, 2019-03-11:
Okay, good!

sbourdeauducq, 2019-03-11:
> Another two go to RTM FPGA SRCC and IO pins

FYI, there is no current requirement to connect SYSREF to CC pins anywhere. But it can't hurt, and we might want extra clocking path if something unexpected happens...

closed 2019-03-11 by sbourdeauducq