sinara-hw/Sayma_AMC #60: Sayma AMC - Version 1 errata

gkasprow, 2019-03-11:
Mechanical:
- [x] use 2mm higher J14 connector.  Wurth 621 010 210 21
- [ ] add cutout in the panel under SFP cage to make module removal easier
- [ ] make 3.5mm switch hole in front panel
- [ ] panel switch hole = 3.5mm
- [ ] sfp cutout 0.5mm bottom
- [ ] make USB connector hole touch panel edge
- [ ] move L23 away from ufl connector
- [ ] Move power supply status LEDS from bottom side of board to top edge of board (like on Sayma_RTM) so that they're visible from top of crate. 

BOM:
- [x] mark R257 DNP
- [ ] add SMA insulating washer to the panel
- [x]  wrong type of guiding socket  - should be "1" instead of '8'
- [x] IC24=DNP
- [ ]  R183=DNP
- [x]  remove C282
- [x] AMC: make C221 0R to enable internal reference
- [x] install R507 by default
- [ ] Include FMC standoffs and screws in BOM. 

I2C:
- [x] connect some I2C pullups to P3V3 instead of P3V3MP

JTAG:
- [x]  add 0R resistors on SWD/JTAG or FET that disconnects SCANSTA during CPU flashing
- [x] add JTAG MPSEL status/control to MMC pin E8
- [x] fix issue with JTAG switchover. After changing MMPSELB1/B0 SCANSTA needs reset to apply settings
- [x] to enable FLASH programming operation, disconnect it from RTM FPGA. Add termination.
- [x] fix second FLASH CSn connection - use pin G26 for CSn instead of K21

USB:
- [x] T23 source = GND
- [x] Insertion/removal of USB cable resets MMC and power supply. Feature: add RC delay circuit or voltage supervisor. Quick fix: jumper on reset signal on T21
- [x] 100k resistor on T23 gate

Ethernet:
- [x] Make sure that TX CLK goes first before Ethernet PHY reset. Feature: trigger it by CONFIG DONE or leave as it is.
- [x] MAX PHY need to have clock delivered to both MII TX CLK and GTXCLK
- [x] make sure that PHY TRSTn is low
- [x] make sure that PHY LOS is low
- [x] Apply Etheret fix #327 

Power:
- [x]  feature: add slow start for P12V FMC and RTM 1k5 + 1uF
- [x] add slow start MOSFET in parallel to the current limiting resistor.
- [x] add 3v3mp LED
- [x] add 12V ext TVS
- [x]  LD11 current too low, R242 = 10k
- [x] increase the VCCINT to 0.975V
- [x] 0.9V generation - R243+R244=5k1
- [x]  swap PSIO2 and GPIO0 - resistors R208, R205

Schematics:
- [x] MAX6642 address is 0x48, not 0x24 - change schematic annotation
- [x]  fix schematic symbol of SATA connector - pins numbering
- [x]  RTM IC40 address is 0x4B - change comment on schematics
- [x] Fix illegible blocks in schematic (cf  #231)
- [x] Zero-index all channels (SMAs, FMCs, SFPs, etc)
- [x] Rename USR_UART_P/N - UART is not a differential signal

Clocks:
- [x] OSC3 = 510BBA200 instead of 510ABA200 or add 2x200R to GND
- [x]  wrong connection of SD pins of LTC6957. They must be tied low instead of high.
- [x] re-route Si5324 clocks so CLK1 and CLK2 have same skew - are connected to clock buffer.

Other: 
- [x] Assign clock capable pins on FMCs
- [x] wrong pin numbering on J13 silkscreen
- [x]  swap T20, T21 drains or gates
- [x] 10k pullup on SFP pin 9
- [x] The SMA_IO1_DIR MOSFET incorrectly controls the SMA_IO2 channel, and vice versa.

RTM:

- [x] replace MAX6642ATT90+T with MAX6642ATT98+T on RTM - change I2C address from 0x48 to 0x4C (already in RTM errata)

hartytp, 2019-03-11:
> There are only mechanical and PCB left

Great. Are the AMC/RTM schematics ready to review then apart from the HMC830 active loop filter OpAmp choice?

hartytp, 2019-03-11:
> I'll move it to Sayma_AMC repo

Thanks. While you're at it, would you mind reviewing all other Sayma related issues in the Sinara repo and closing/moving them as appropriate?

closed 2019-03-25 by gkasprow