sinara-hw/Sayma_AMC #63: consider making all UFLs MMCX

hartytp, 2019-03-13:
It would be easier if I didn't have to stock two types of cable. If there are layout/mechanical reasons why this isn't easy for all connectors then it's fine to leave things how they are.

sbourdeauducq, 2019-03-22:
uFLs are lower-profile and make jumper cables slightly easier to use e.g. if the board has to be put into the uTCA chassis with additional jumper cables installed (as is necessary now for DAC synch on v1).

sbourdeauducq, 2019-03-22:
For example, it still fits in the uTCA crate with uFL jumper cables on the bottom side of the boards.

gkasprow, 2019-03-25:
I placed all uFL connectors on the bottom of the board because of clock distribution circuits resites there

closed 2019-03-25 by gkasprow