sinara-hw/Sayma_AMC #65: DFF decoupling

hartytp, 2019-03-13:
Is it worth sticking 100nF at each VBB/VCC pin on the DFF instead of one for the whole package?

hartytp, 2019-03-13:
I won't check the LVPECL to SSTL conversion, but please do make sure you're happy with it and that we get a decent signal at the FPGA.

gkasprow, 2019-03-25:
There is one VBB pin. AD library instantiates it twice. The same applies to VCC pins that normally are hidden but AD19 do not really support hidden supply pins so I had to enable them and connect to desired power net. So treat it as a temporary solution, all CERN libraries would have to be modified to support AD19.
The DIFF SSTL clock entry would work fine.  The SSTL18 single-ended clock entry does not have such high margin but should work. Diff SSTL15 requires 100mV of clock amplitude while SSTL18 requires 125mV and we are driving it single ended and are on the edge of the specification in extreme temperature cases.

closed 2019-03-25 by gkasprow