sinara-hw/Sayma_AMC #68: Sayma buck converter

gkasprow, 2019-03-18:
I want to kick out the Exar option completely, not only DNP. It messes up with the layout.
I have a bunch of tiny modules I developed for another project, roughly 2x2x1cm with Exar 7724 chip on it. I will just add mounting holes for these modules as a backup option.

jbqubit, 2019-03-19:
Please clarify what you mean. You're kicking out Exar but then using Exar 7724 on module? What's the role of what's soldered on the board vs the modules you're proposing?

hartytp, 2019-03-19:
> Please clarify what you mean. You're kicking out Exar but then using Exar 7724 on module?

I believe that what he means is that he's kicking the Exar out completely. But, as a risk mitigation strategy, adding some pads so that we could add a small board with a different regulator (such as the Exar chip). It's just paranoia to cover the (hopefully unlikely) scenario that the new regulator works less well than the old exar.

gkasprow, 2019-03-19:
Exactly. Just risk mitigation. I didn't have any issues with these Exar7724 modules so will leave it as an option in case we have issues with the module.

gkasprow, 2019-03-25:
I added a few mounting holes for the module. Below I fitted several components so no PCB real estate is wasted by DNPs
[image]

closed 2019-03-25 by gkasprow