sinara-hw/Sayma_AMC #70: "Si5325" typo

sbourdeauducq, 2019-03-21:
[image]

sbourdeauducq, 2019-03-21:
Also, why is that a "WR clock input"? What are the exact use cases for this connector and signal?

hartytp, 2019-03-21:
@sbourdeauducq It's not really a WR clock input at all. IIRC, it's just an unbuffered connector that goes to a GC pin.

The main use WR case I can think of for WR would be the reference clock if Sayma were used as the master. However, I suspect that it will be useful for testing and debugging the WR implementation. But, I would support a better name for the signal.

sbourdeauducq, 2019-03-22:
Let's just call it "AMC external clock input".

closed 2019-03-25 by gkasprow