sinara-hw/Sayma_AMC #71: 32-bit DDR DQS P/N swapped

sbourdeauducq, 2019-03-21:
The DQS_P signals go on the N side of the I/O buffers, and vice versa.
Was this done to ease board routing?
If really necessary, we can deal with it with the MiSoC controller, but it makes things a bit unusual and I don't know if the Xilinx cores (and others) will be happy with that.

gkasprow, 2019-03-21:
It happened probably during design migration. Good catch!
I will run SDRAM simulator anyway after I finish routing.

marmeladapk, 2019-03-22:
@gkasprow @sbourdeauducq Whole IC37I is swapped upside down (mirrored along X axis). This is the real issue here. This also means that VRP signal is not connected where it should be.

gkasprow, 2019-03-22:
OK, I'll fix it.

gkasprow, 2019-03-22:
I'm a bit surprised that with mirrored connections the stub port did not generate serious issues.
DDR3 controller uses DQS pins in a quite restricted way.

sbourdeauducq, 2019-03-23:
> DDR3 controller uses DQS pins in a quite restricted way.

In MiSoC, we're not using the "native mode I/O" where most of the DQS restrictions come from.

> I will run SDRAM simulator anyway after I finish routing.

Does that include trying to compile a design with the Xilinx SDRAM core?

gkasprow, 2019-03-24:
OK, fixed.
I did not touch traces that were routed in rev1. And noticed that Altium had problems with proper sync between sch and pcb. With fixed schematic the problems disapperared.
I will redo verification with Hyperlynx SiDDR tool. It check SI and crosstalk only, it's verilog model does not take into account Xilinx IP.

closed 2019-03-25 by gkasprow