sinara-hw/Sayma_AMC #79: DRC rules & DRC

marmeladapk, 2019-03-29:
- Power pin connected to output should generate at most a warning (most bucks and LDO's have voltage output marked as an output pin, not as a power pin)
- Output pin and bidirectional sheet entry (and port) shouldn't create an error (output pin and bidirectional pin don't create errors)
- generally I feel that output or input object shouldn't create an error with bidirectional object

Small issues found:
- LTM_PGOOD port on QUAD_BUCK schematic should be an output port
- PGOOD port on SDRAM_DDR3_2x16 should be an output
- T29B should probably be shown on schematic with all pins NCd

All other errors (and one fatal error) should be suppressed using NO ERC mark which targets a specific DRC.

I looked through most of other warnings. Multiple net names could probably be resolved one day, I managed to do this somehow on Kasli, but it requires a lot of experimentation, not worth it right now.

Please close this issue when there are no error during compilation.

gkasprow, 2019-04-26:
if we make PGOOD output port, Altium won't like it either.

gkasprow, 2019-05-20:
I want to leave the warnings. If we place "no ERC" and if there is some issue on that line, it won't find it.

closed 2019-05-20 by gkasprow