sinara-hw/Sayma_AMC #104: Autorouter madness

marmeladapk, 2019-04-15:
[image]

marmeladapk, 2019-04-15:
[image]

marmeladapk, 2019-04-15:
[image]
Shared via

marmeladapk, 2019-04-15:
[image]

marmeladapk, 2019-04-15:
IC32A

[image]

marmeladapk, 2019-04-15:
Near IC21

[image]

marmeladapk, 2019-04-15:
Near C441
[image]

marmeladapk, 2019-04-15:
General remark: more vias to ground for caps on the right of the FPGA?

marmeladapk, 2019-04-15:
Q1

[image]

marmeladapk, 2019-04-15:
Si5324 could use moure grounding vias

marmeladapk, 2019-04-15:
C102
[image]

marmeladapk, 2019-04-15:
3 capacitors share the same ground via near C355

marmeladapk, 2019-04-15:
[image]

Not needed wobbles

gkasprow, 2019-04-27:
some loops are intentional. The first one is to protect traces agaisnt ESD discharge.

closed 2019-04-27 by gkasprow