sinara-hw/Sayma_AMC #113: Power budget (AMC+RTM)

kaolpr, 2019-04-16:
There seems to be an issue with mTCA compliance regarding total power on AMC and RTM. AMC and RTM may require up to 45W and 95-100W, respectively. It sums up to 140-145W, whereas mTCA specification states that:

- The average power on the uRTM **shall not** exceed 30 W.
- The maximum power available to the AMC/RTM combination is 80W.

gkasprow, 2019-05-20:
The power consumption estimation was under the assumption that we fully load all AFE rails, FMC rails and run all FPGA with full speed.  Currently, the full set of AMC + RTM + AFEs consumes roughly 40..50W of power. So we still have 30..40W of margin.

closed 2019-05-20 by gkasprow

marmeladapk, 2019-07-16:
[image]

🤔

Either way, 4,5A on 1V5 is still concerning, since LTM4644 can output 4 A.

Found by @tprzywoz

gkasprow, 2019-07-24:
Well, this value was assumed with a large margin, taken from max power the memory can dissipate.