sinara-hw/Sayma_AMC #115: Sayma AMC SI verification

gkasprow, 2019-05-02:
Results of DDR verification
[image]
[image]
[image]
[image]

gkasprow, 2019-05-02:
[image]
[image]
[image]
[image]

gkasprow, 2019-05-02:
[image]
[image]
[image]
[image]
[image]

gkasprow, 2019-05-02:
[image]
[image]

gkasprow, 2019-05-03:
The whole simulation including crosstalks and FR4 losses takes roughly 12hours on quad Xeon machine. Now I'm simulating DDR32 memory.

gkasprow, 2019-05-03:
@sbourdeauducq I'm doing simulations for 800MT/s (400MHz clock) Is that sufficient?

gkasprow, 2019-05-03:
DDR32 results
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

sbourdeauducq, 2019-05-04:
>  I'm doing simulations for 800MT/s (400MHz clock) Is that sufficient?

The current ARTIQ/MiSoC does 1000MT/s (and we might want to crank that up in the future).

gkasprow, 2019-05-04:
OK, so I will optimize it for 1066MT/s

closed 2019-05-20 by gkasprow