sinara-hw/Sayma_AMC #132: jumper to disable microTCA hotplug switch

sbourdeauducq, 2019-10-04:
Since Creotech are shipping boards without front panels (which are delayed), add a jumper to force the µTCA hotplug switch on. Putting tape on it to turn it on is annoying and unreliable.
Same on RTM.

marmeladapk, 2019-10-05:
This is a one-off issue. Next batch will have panels already. Besides it's easier to force power on in MMC (or even use the front panel button). Jumper isn't needed IMO.

sbourdeauducq, 2019-10-06:
> Besides it's easier to force power on in MMC

Does this involve IPMI or NATview? If so, it's not easier than a jumper.
Is the front panel button currently wired to force power to both AMC and RTM? This does not sound like such a great idea as users may inadvertently push it while hotplugging boards.

kaolpr, 2019-10-06:
No, it does not involve IPMI / NATview. What @marmeladapk proposed is modification to AMC MMC, which manages RTM power supply. We can than use AMC reset button to signalize MMC that RTM / AMC is to be switched on/off. Please note that this is **temporary workaround** until panels are produced. Currently, there is very limited number of users (actually, besides Creotech you are the only one), so we believe that exposition on inadvertent actions is minimal.

marmeladapk, 2019-10-06:
Or you could turn it on and off using mmc console simulating hotswap position to MMC.

sbourdeauducq, 2019-10-16:
Any update on those front panels?
Also, how do I force power on with the MMC console?

marmeladapk, 2019-10-16:
@sbourdeauducq We should get all front panels on the last week of October.

> how do I force power on with the MMC console?

Currently you can't, we'd have to add this to MMC console. Is this critical for you now?

sbourdeauducq, 2019-10-16:
Not critical but replacing the tape on the switches every few days is annoying. I guess I could also solder them short...

sbourdeauducq, 2019-12-15:
@marmeladapk Any update on the front panels or MMC firmware with that feature?

sbourdeauducq, 2019-12-21:
> MMC firmware with that feature

Nevermind, I soldered the switches short.

As @jbqubit reminded you, please create a wiki page listing all boards and where one can log reworks made to them.