sinara-hw/Sayma_AMC #133: xlsx/schematics mismatch

sbourdeauducq, 2019-10-04:
The .xlsx file with the FPGA pin assignments for v2.0rc8 (production) does not match the schematics.

This shows FPGA pin E28 assigned to DDR3_32_DQS3_N:
https://github.com/sinara-hw/Sayma_AMC/releases/download/v2.0rc8/Sayma_AMC_FPGA_pins.xlsx 

This shows FPGA pin E28 assigned to DDR3_32_DQ5:
https://github.com/sinara-hw/Sayma_AMC/releases/download/v2.0rc8/Sayma_AMC.PDF

Which one corresponds to the boards that have been produced?

@gkasprow Can you PLEASE be VERY careful about this sort of thing? This recurrent pattern has been causing us a lot of problems.

gkasprow, 2019-10-04:
I traced back what happened.
After I moved from EE to ALtium, the translator mirrored the schematic symbol. S, all DDR32 pins are mirrored in the xlx file.
During routing I noticed that and fixed.
The xls file is very old (march 2019). I forgotten to generate the new one.
I updated the xls file in the release
https://github.com/sinara-hw/Sayma_AMC/releases/tag/v2.0.1

closed 2019-10-04 by gkasprow

sbourdeauducq, 2019-10-04:
Is this 2.0.1 tag still applicable to the v2 boards that have been already produced?
What are the differences?

gkasprow, 2019-10-04:
Yes, only xls file was not updated.

marmeladapk, 2019-10-06:
@sbourdeauducq https://github.com/sinara-hw/Sayma_AMC/commit/75b020da3583e4195b526ac2c1a2b15560f89bde