sinara-hw/Sayma_AMC #136: detection of RTM hotplugging?

sbourdeauducq, 2019-10-08:
How to detect from the AMC that the RTM is being unplugged/re-plugged?
Now that the AMC-RTM link is using DRTIO, we can support RTM hotplug without a lot of trouble; we just need to reload the RTM FPGA when the RTM is plugged in and make sure that the JESD core is properly reset.

kaolpr, 2019-10-08:
Hotplug switch on RTM is connected to the GPIO expander, which is connected to AMC I2C subsystem. Thus you can access it using AMC MMC or FPGA I2C master.

jbqubit, 2020-01-17:
Any luck with this @sbourdeauducq ?

marmeladapk, 2020-01-27:
@sbourdeauducq We detect RTM using MMC and it uses FPGA_STATUS to signal this to AMC FPGA. High if RTM is detected and has enabled power, low otherwise. But first we must fix https://github.com/sinara-hw/Sayma_AMC/issues/145.

sbourdeauducq, 2020-01-27:
OK, sounds good.