sinara-hw/Sayma_AMC #139: WR FF to FPGA connection

sbourdeauducq, 2019-12-07:
Why are there both single-ended SSTL18 connections and differential SSTL15?
Which one is populated on the boards?

Also this indication of "LVDS" is misleading:
[image]

@gkasprow

gkasprow, 2019-12-07:
They are just in case. There is a note below that explains this.
LVDS is just a routing rule.
At the moment the single-ended SSTL18 is populated, so do not use the differential ones.
It is a source terminated.

sbourdeauducq, 2019-12-08:
Currently I cannot get any signal out of those FFs.
The FPGA constraints I am using are:
```
Subsignal("rec_clk", Pins("F12"), IOStandard("SSTL18_I")),
Subsignal("main_xo", Pins("L9"), IOStandard("SSTL18_I"))
```
Are they correct?

gkasprow, 2019-12-08:
They look fine.
Please, look with fast scope at the pins 14, 15 and 16, 17 of IC19.
If you don't have such fast scopes, measure DC voltages instead.

gkasprow, 2019-12-08:
@marmeladapk did you check this circuit?

marmeladapk, 2019-12-09:
@sbourdeauducq We checked `DDMTD_OUT_MAIN_CLK_SSTL18`. And the signal that we got out of it looked fine. We used the same IO standard as you. Do you have both main and helper clock signals?

With CDR we had a problem to get clock out of MGT. It didn't work on the first few tries, so we moved to other issues.

sbourdeauducq, 2019-12-09:
I have both working now (since today).
The GTH needs this: https://github.com/m-labs/artiq/commit/307a6ca140b1cea8cf0c08072fb733312de6fd0a

sbourdeauducq, 2019-12-09:
Another issue was this: https://github.com/m-labs/artiq/commit/14e09582b6ea141624d3f50d85498d4590f55c7c

marmeladapk, 2019-12-20:
@sbourdeauducq Can we close this?

sbourdeauducq, 2019-12-20:
Just two issues remain:
1. While I am getting consistent DDMTD data from the main XO FF, the DDMTD data from the GTH FF is unexplained. I do not know where the problem is coming from. FWIW, I am also getting inconsistent data with the RTM implementation that does not use FFs, so it is perhaps not the FFs.
2. If we don't use the differential signal, it should perhaps be removed.

jbqubit, 2019-12-20:
I created separate issue about the misleading LVDS label. https://github.com/sinara-hw/Sayma_AMC/issues/143

jbqubit, 2020-01-17:
@sbourdeauducq Can this be closed?

sbourdeauducq, 2020-01-17:
No, there should still be a decision about which signal (SE or differential) to keep, and then remove the other.