sinara-hw/Sayma_AMC #140: LVDS termination for signals received from Si549s

sbourdeauducq, 2019-12-08:
Should the FPGAs enable integrated LVDS terminations (DIFF_TERM=TRUE) for those signals that it receives from the DCXOs?

I suppose that should use the LVDS standard even when the DCXO is CMOS, so that LVDS DCXOs can be used instead without changing the bitstream.

(Same question for AMC and RTM)

gkasprow, 2019-12-08:
Actually, the CMOS oscillators are not stocked and we will use LVDS ones. At least in Metlino.
You can use either 2x SSTL18 or LVDS. The latter consumes less current.

sbourdeauducq, 2019-12-08:
Ok, and should there be DIFF_TERM=TRUE or not? And I was asking about Sayma v2 (the boards that have already been produced) and Kasli v2.

jbqubit, 2020-01-17:
How was this resolved?