sinara-hw/Sayma_AMC #142: BaseMod RF, output amplitude inconsistent

jbqubit, 2019-12-19:
Issue raised on call with Creotech on 12/18. "RF output has correct frequency but every second channel has 2x amplitude." @marmeladapk says this might be due to error in BaseMod configuration. 

@sbourdeauducq Has M-Labs written a driver for BaseMod yet? I gather Cretoech would appreciate using your driver as a reference design.

gkasprow, 2019-12-19:
Unitinialised attenuators often setup in a random way, I noticed it on Urukul.

marmeladapk, 2019-12-20:
@jbqubit It not an error per se in configuration, it's just lack of configuration ;)

marmeladapk, 2019-12-20:
@atcher0 fixed it yesterday before we had a chance to tell him that it's not needed.

closed 2019-12-20 by marmeladapk

sbourdeauducq, 2019-12-20:
FWIW, the HMC542 register says that reset held low "clears the shift register".
I do not know if reset affects the latch (I suppose we have to measure that ourselves, as is often the case when using anything from HMC) and thus if a pull-down resistor on reset would help with this minor wrinkle.

sbourdeauducq, 2019-12-20:
The schematic says it does not affect the latch:
[image]

So it would be pull-down on reset and pull-up on LE.

gkasprow, 2019-12-20:
Yes, but first, someone must toggle the reset line :)

sbourdeauducq, 2019-12-20:
According to the datasheet, reset is level-sensitive. But we would have that problem with the latch signal, yes (which is rising edge sensitive).
[image]

sbourdeauducq, 2019-12-20:
So if we really care about this, we need some complicated power-up sequencing of LE and reset pullup/pulldowns. HMC design strikes again...
But since there are also RF switches on our PCB, it doesn't really matter.

sbourdeauducq, 2019-12-20:
And it's not a "latch" but a D flip-flop...

sbourdeauducq, 2019-12-21:
> @sbourdeauducq Has M-Labs written a driver for BaseMod yet? I gather Cretoech would appreciate using your driver as a reference design.

Yes and it seems to work - at least I can read back the values.
See the latest commits in the ARTIQ repos.