sinara-hw/Sayma_AMC #144: flimsy µTCA hotplug switches on NAT modules

sbourdeauducq, 2019-12-21:
The hotplug switches on NAT modules (NAT-MCH and AC-600D power supply) are unreliable. 

Sometimes the crate shuts down because a slight vibration of the table caused a glitch in the switch and detection of a hotplug event. Or sometimes, the hotplug handle, even though it is fully inserted, needs to be wiggled a bit for the crate to start.

Is this a known problem and what is the solution?

jbqubit, 2019-12-24:
@vdirksen and @gkasprow might have thoughts on this.

kaolpr, 2019-12-24:
Unfortunately @vdirksen passed away this year.

marmeladapk, 2020-01-17:
One of @jbqubit 's MCH also exhibits this behaviour.

jbqubit, 2020-01-22:
I've sent notice about this and several other Issues to support@nateurope.com.

jbqubit, 2020-01-27:
Response from NAT is below. It looks like they didn't even read this Issue. I'll encourage them to use github. 

> There should be no principle problem with the hotswap handles. Does this happen with one specific module, one module type (PM, MCH) or with multiple? Do you see any unusual mechanical behavior of the handle?