sinara-hw/Sayma_AMC #145: RTM hotswap

marmeladapk, 2020-01-10:
### Problems

- When RTM power is disabled, FPGA unprogrammed
  - PDONE LED is on, however when I touch the gate of transistor with oscilloscope probe it turns off, I measure around 450 mV
  - P3V3 has 2,7V

### Signals between AMC and RTM FPGAs and ICs

- RTM UART - 400 mV
- RTM_FPGA_CFG
  - INIT_B 1,4V
  - Done 400mV (but LED shuts off so measurement changes state)
  - Program_b 400mV
  - CCLK 400mV
  - DIN 400mV
- RTM I2C
  - 3,3V and talks (however it's connected to 3V3MP on AMC)
  - TMP_SENS is pulled up to P3V3 and has 2,8V (but it could be from another source, MMC doesn't seem to be talking there)
- RTM_JTAG - potential cause, has pullups, however without USB plugged in 3V3 on RTM is also high
  - **TMS 3,3V**
  - **TDI 3,3V**
  - TDO 1,7V
  - **TCK 3,3V**
    - AMC: PROG_TCK 3,2V
  - AMC has internal resistive pullup on these pins
- FPGA_LINK GTP

### Fixes for ARTIQ

- rtm_fpga_cfg -> done pin - shouldn't have pullup - DONE signal already has internal pullup in RTM FPGA and sets signal high before RTM FPGA has power
- init_b, program_B shouldn't have pullups as well

### Removing IC15 & IC13

- When I removed IC15 P3V3 voltage dropped to 2,5V from 2,7V when RTM was not powered
- When I removed IC13 & IC15 P3V3 voltage dropped to 0V + PDONE is no longer illuminated
- Only P3V3MP LED is illuminated
- Pressing hotswap turns on all supplies as expected

So IC15 and IC13 shouldn't supply high level signal to RTM when RTM is disabled.

marmeladapk, 2020-01-15:
Possible ways to move forward with current boards:
- abandon hotswap on RTM until v2.1
- rework AMC boards
    - remove IC13, IC14
    - populate R59-R62 (JTAG bypass)
    - this will remove RTM and FMC from JTAG chain

Thoughts @jbqubit, @gkasprow ?

gkasprow, 2020-01-15:
We don't need hot-swap anyway, do we? Software does not support it.

jbqubit, 2020-01-15:
RTM hotswap is a low-priority feature. If skipping hotswap for now helps things move forward faster let's do that.