sinara-hw/Sayma_AMC #146: No power to some cards in crate

marmeladapk, 2020-01-17:
- Multiple Sayma cards in R5 crate (crate power turned off)
- NAT MCH
- All AMCs have inserted hotswap

When turning on crate power, MCH boots up and grants payload power to 1-2 boards (always in the same slots).

jbqubit, 2020-01-22:
@marmeladapk Please post your MCH configuration file so NAT Support can take a look.

marmeladapk, 2020-01-27:
Same as with https://github.com/sinara-hw/Sayma_AMC/issues/147

One other thing that changed was switching back to debug compilation instead of release.

jbqubit, 2020-01-27:
This isn't really a satisfactory resolution. Which device are you compiling for as regards debug/release?  NAT's response is the following. 

> Can you please provide a full system information log file of the system? You can create the log archive via the web interface of the MCH.