sinara-hw/Sayma_AMC #148: confusing SYNCOUT connections

sbourdeauducq, 2020-01-20:
Are they on LVDS3/LVDS11 or on LVDS28/LVDS29?

[image]

[image]

kaolpr, 2020-01-20:
Following the signal: LVDS29 on RTM is LVDS29 on AMC and its connected to J13/H13. 
Label near LVDS3/LVDS11 is misleading and should be moved to LVDS28/LVDS29.

sbourdeauducq, 2020-01-20:
> Following the signal: LVDS29 on RTM is LVDS29 on AMC and its connected to J13/H13.

What about the other SYNCOUT? We're seeing weird issues on only one DAC so let's check this carefully.

kaolpr, 2020-01-20:
SYNCOUT10 -> AMC / LVDS28
SYNCOUT20 -> AMC / LVDS29