sinara-hw/Sayma_AMC #150: AMC photos

jbqubit, 2020-03-31:
Please post photos of the stuffed Sayma_AMC for contract accounting. 

https://github.com/sinara-hw/Sayma_AMC/wiki/Sayma_v2_contract_creotech#c332-ht3-manufacturing-baseline-testing

Sayma AMC v2.0: print 10 PCBs, stuff 6

akaminska, 2020-04-07:
1907180PW0009A0103
[image]
[image]

akaminska, 2020-04-07:
1907180PW0009A0104
[image]
[image]

akaminska, 2020-04-07:
1907180PW0009A0106
[image]
[image]

akaminska, 2020-04-07:
1907180PW0009A0105
[image]
[image]

akaminska, 2020-04-07:
1907180PW0009A0107
[image]
[image]

akaminska, 2020-04-07:
1907180PW0009A0108
[image]
[image]

akaminska, 2020-04-07:
Unpopulated PCBs
[image]
[image]

jbqubit, 2020-04-07:
Thank you. Please also post photo of at least one AMC with front panel installed. Ideally front view.

akaminska, 2020-04-10:
Please remember that these panels are "prototypes" as well, we definitely aim for better quality for the final product...
[image]
[image]

closed 2020-04-10 by jbqubit